PlummersSoftwareLLC/NightDriverStrip #633: m5plusdemo blank screen

phpbbireland, 2024-06-25:
**Problem**

**Steps**

1.  Set default_envs = m5plusdemo
2.  build completes without errors
3.  upload completes without errors

**Notes**

Any suggestions on how I should proceed?

rbergen, 2024-06-25:
@phpbbireland I think we're going to need a little more information before we can say anything useful. Like:

- Exactly what device are you flashing to?
- Which "screen" is black/blank?
- What's the serial console logging that the device outputs while booting?

And as far as I can tell by doing a search in the codebase, User_Select_Setup.h is not part of any source file in this project.

phpbbireland, 2024-06-25:
Device: M5 Stick C Plus 1.1 (M5StickC PLUS ESP32-PICO Mini IoT Development Kit)...
I have not connected the LED's yet, the screen on the M5 Stick is blank...

I guess I should have looked at the console... 
Console:
`E (657) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
E (634) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.
assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)
`

rbergen, 2024-06-26:
That's a fault condition I'm personally not familiar with. The "binary image header" size of 16 megs is clearly too big, though.

rbergen, 2024-06-26:
I think I see what the problem is. The m5plusdemo config in platformio.ini is actually set up to build for the M5Stack Core2. Looking at the commit history, Dave reused the m5plusdemo config to test the work he was doing towards adding support for the M5Stack Core2 to the project.

To get the config to work on the M5StickC Plus again, you can try changing the following section in platformio.ini:

```ini
[env:m5plusdemo]
extends         = dev_m5stack
build_flags     = -DM5DEMO=1
                  ${dev_m5stack.build_flags}
```

to 

```ini
[env:m5plusdemo]
extends         = dev_m5stick-c-plus
build_flags     = -DM5DEMO=1
                  ${dev_m5stick-c-plus.build_flags}
```

and then building, flashing and booting again.

phpbbireland, 2024-06-26:
Excellent... that was the issue, thank you very much...

I had little hope of fixing this, thanks again... now to play ;)

closed 2024-06-26 by phpbbireland

rbergen, 2024-06-26:
That's good to hear!

I've opened #634 to "restore" the m5plusdemo config to target the M5StickC Plus, and add a separate m5stackdemo config for the M5Stack Core2. That should avoid this confusion in the future.

In the meantime, have fun playing! 🙂

PR #634 (2024-06-26): Make demo config available for M5StickC Plus and M5Stack Core2
## Description

This makes the m5plusdemo config build for the M5StickC Plus once again, and adds a separate m5stackdemo config that targets the M5Stack Core2. Fixes #633.

## Contributing requirements

* [x] I read the contribution guidelines in [CONTRIBUTING.md](http://github.com/PlummersSoftwareLLC/NightDriverStrip/blob/main/CONTRIBUTING.md).
* [x] I understand the BlinkenPerBit metric, and maximized it in this PR.
* [x] I selected `main` as the target branch.
* [x] All code herein is subjected to the license terms in [COPYING.t [... cut]