PlummersSoftwareLLC/NightDriverStrip #640: stock effect triggers error in the serial

heidepiek, 2024-07-04:
This week I downloaded the latest version of the code from the nightdriver github site. Was curious if some new stuff had been added. And saw that a stock effect had been added. Now I have nothing to do with stocks myself, but autistic as I am. If something is added it should work. But when I click the stocks effect I get the following message in the serial: 

(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65276, LargestBlk: 31732, PSRAM Free: 387827/4172235, LED FPS: 25 Refresh: 60 Hz, Power: 2426 mW, Brite: 100%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 050%, 013%, FreeDraw: 0.035
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(I) (WriteCurrentEffectIndexFile)(C0) Number of bytes written to file /current.cfg: 2
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65276, LargestBlk: 31732, PSRAM Free: 387827/4172235, LED FPS: 30 Refresh: 60 Hz, Power: 4641 mW, Brite:  91%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 065%, 044%, FreeDraw: 0.017
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65276, LargestBlk: 31732, PSRAM Free: 387827/4172235, LED FPS: 30 Refresh: 60 Hz, Power: 4641 mW, Brite:  97%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 051%, 035%, FreeDraw: 0.024
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65276, LargestBlk: 31732, PSRAM Free: 387827/4172235, LED FPS: 30 Refresh: 60 Hz, Power: 4641 mW, Brite:  97%, Audio FPS: 47, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 052%, 034%, FreeDraw: 0.027
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65276, LargestBlk: 31732, PSRAM Free: 387827/4172235, LED FPS: 30 Refresh: 60 Hz, Power: 4641 mW, Brite:  97%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 050%, 034%, FreeDraw: 0.018
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65276, LargestBlk: 31732, PSRAM Free: 387827/4172235, LED FPS: 30 Refresh: 60 Hz, Power: 4641 mW, Brite:  97%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 053%, 041%, FreeDraw: 0.023
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65276, LargestBlk: 31732, PSRAM Free: 387827/4172235, LED FPS: 30 Refresh: 60 Hz, Power: 4641 mW, Brite:  97%, Audio FPS: 50, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, CPU: 051%, 033%, FreeDraw: 0.023
(W) (NotifyJSONWriterThread)(C1) >> Notifying JSON Writer Thread
(W) (NotifyNetworkThread)(C1) >> Notifying Network Thread
(I) (FetchQuotes)(C1) Starting update of stocks...
[ 71075][E][WiFiClient.cpp:268] connect(): socket error on fd 54, errno: 104, "Connection reset by peer"
(E) (GetQuote)(C1) [HTTP] GET failed, error: connection refused
(E)
[ 71270][E][WiFiClient.cpp:268] connect(): socket error on fd 54, errno: 104, "Connection reset by peer"
(E) (GetQuote)(C1) [HTTP] GET failed, error: connection refused
(E)
[ 71476][E][WiFiClient.cpp:268] connect(): socket error on fd 54, errno: 104, "Connection reset by peer"
(E) (GetQuote)(C1) [HTTP] GET failed, error: connection refused
(E)
[ 71782][E][WiFiClient.cpp:268] connect(): socket error on fd 54, errno: 104, "Connection reset by peer"
(E) (GetQuote)(C1) [HTTP] GET failed, error: connection refused
(E)
(I) (loop)(C1) WiFi: WL_CONNECTED, IP: 192.168.1.248, Mem: 65120, LargestBlk: 31732, PSRAM Free: 387579/4172219, LED FPS: 25 Refresh: 60 Hz, Power: 2926 mW, Brite: 100%, Audio FPS: 52, MinVU:    0.0, PeakVU:    0.0, VURatio: 0.0 Buffer: 0/500, 

Is there anything I'm overlooking? Is there anything that needs to be added in a file? Know I can also turn off the effect, but what I said. If the option is there, it's supposed to do it in my head. 

[image]

Regards
Harrie

davepl, 2024-07-04:
Stock server may not have been running.  Try it now.

- Dave

davepl, 2024-07-04:
BTW you can verify that the stock server is running as it should be at:

http://davepl.com:8888/?ticker=MSFT￼
/
Text Document · 4 KB

heidepiek, 2024-07-05:
hi dave, 
when I click on this link I get the following on the screen:

[image]

But the stock effect does work on mesmerizer now!
Don't know why the link doesn't work, but mesmerizer can do something with it by the looks of it. Maybe something to work on?

Now I have to hurry to work, I'm late again. But had to try this! 

Regards 
Harrie

rbergen, 2024-07-05:
That's because depending on where/how you use the link, there are some bogus characters at the end of it. If you remove everything after MSFT from the URL, it will work.

(Actually, the "no data available" message already proves the required service is running, but it isn't its most interesting output. :) )

rbergen, 2024-07-05:
I've now edited Dave's comment to fix the link. It should "just work" now.

heidepiek, 2024-07-05:
Thanks dave and rutger!  I think dave was right that the stock server was not running. (wednesday, juli 3) 
I wasn`t the only one that could not get data for the stocks effect.  But just got home, and turned on the mesmerizer and It works properly. The bogus characters i did not see when i read dave`s reply 🙈 

Wishing everyone a great weekend! 👊 

Greetz

Harrie

closed 2024-07-05 by rbergen

robertlipe, 2026-01-03:
> <PlummersSoftwareLLC/NightDriverStrip/issue/640/issue_event/13408329812@
> github.com>

I don't know how we feel about bumping closed issues, but I'd bet that the
cause and the needed fix are the same.

PeakVU:    0.4, VURatio: 0.0 Buffer: 0/500, CPU: 086%, 087%, FreeDraw: 0.022
(E) (GetQuote)(C1) [HTTP] GET failed, error: connection lost
(E)
(E) (GetQuote)(C1) [HTTP] GET failed, error: connection lost
(E)
(E) (GetQuote)(C1) [HTTP] GET failed, error: connection lost
(E)
(I) (UpdateSubscribers)(C1) Got YouTube subscriber count for channel Daves

The test case offered above:
http://davepl.com:8888/?ticker=MSFT
This site can’t be reached

The connection was reset.

(.venv) ➜  nightdriverstrip git:(feature/fuzzy-effect-selection) ✗ curl "
http://davepl.com:8888/\?ticker\=MSFT"

curl: (56) Recv failure: Connection reset by peer

I think all signs point to this code trying to open an endpoint that's not
there. ISTR that this has come up a few times.It seems the owner of davepl
doesn't want to be in the stock quote business (I wouldn't either...) as
they don't have endpoint monitoring, and there aren't very many people
using this effect because people don't notice when it goes down. (it's been
down at least a week and nobody's reported it.)

Is it time to move to another endpoint that provides similar data? There
are many places that provide the data, but even for personal use,
it requires signup and a key. I'm not going to list them and instead
suggest a slightly different attack.

If we push people to sign up for an API key - as we already do for
OpenWeather - what if we instead direct people to a clearinghouse that
provides a ton of different data that might be interesting for
Mesmerizer-like displays?

One such is rapidapi.com They have a LOT <https://rapidapi.com/collections>
of such APs. Some are free
<https://rapidapi.com/collection/list-of-free-apis>. Most have usage
limits that can be raised in exchange for beaver pelts or precious metal
bullion. Some are expensive
<https://rapidapi.com/therundown/api/therundown/pricing> in ways I don't
really understand.

They have a ton of stock ticker services. Most of them are freemium and
offer graduated levels of service. Financebird, for example offers free to
3,500 quotes per month. The APIs are straight-forward and can return JSON
that looks like we can munch.

Are there enough users of this service/effect to justify rewiring it to a
different provider?

Now, the "why rapidapi?" question...

One of our oldest requests was for sports scores. They have sports
<https://rapidapi.com/search?term=sports&sortBy=ByRelevance>.

One could imagine base classes that build requests to rapidjson, manage
timeouts, restrict excessive calling, issue callbacks with the returned
objects created/modified by the API call.

So for this issue, we have a two parter:
1) Should davepl keep fielding 'please press the reset button' requests
when someone notices that it's failing? (Answers for that probably best
come from Davel...
2) Should we move this dependency to something else? If yes, does something
like financebird
<https://rapidapi.com/shareefbassam3/api/financebird/pricing> (3,500 free
requests/month) work for the users of this? That's 116 requests/day. Limit
it to business hours - and figure out what that means for various
timezones - and that's ... reasonable--sounding.

P.S. If not rapidapi for sports, does something like:
https://www.thesportsdb.com/documentation
work for those wanting sports?

rbergen, 2026-01-03:
We don't have a policy on bumping closed issues, but there is a meaningful risk of bumps being missed if the issue in question doesn't show up in the open issue list. (I.e., a bump may be best combined with a reopen of the closed issue.)

Your comment touches on a few points:

- The stock server proxy that the Stocks effect is connecting to was functionally down. I've fixed that by restarting it. I've updated the script that checks if the container is running to restart the proxy server every day at midnight PST. I'll try to remember to keep an eye on the container in question to see if this is enough for a somewhat reliable operation.
- The hostname for the stock server proxy has changed quite a long time ago from davepl.com to davepl.dyndns.org; this has since then also been the default value for the effect setting in question.
- The stock server proxy connects to one of the many services that provide stock data. The reason the stock server proxy exists is to convert the default output of the stock data service to a format that is more effect-friendly, and publish it over plaintext http - the http client we use does not include support for TLS. This is something all other stock data services that were looked at at the time required.
- The sports effect doesn't exist yet, and figuring out the service to use for it would be part of developing it. Whoever decides to pick up that gauntlet can incorporate your ideas/suggestions if they want to (and find this exchange of comments, of course). Personally, I wouldn't be surprised if a proxy server similar to the one developed for the stock data would be required.

rbergen, 2026-01-03:
The "you" is plural, in the sense that one of Dave's machines runs it, and I restarted it and put the daily restart cron job in place this morning. The host machine now also runs another script every 5 minutes that checks if the container is running at all. That's not the same as testing if the API is actually functional, but any proxy container related outage should now last no more than 24 hours.

Let's see how we fare with this setup for now. If this turns out to still be too shaky in the practical sense then I'll dive into it further.

robertlipe, 2026-01-03:
Awesome.

I agree. Given the number of people that notice and report the issues,
let's try repair over replacement first.

Thanx for defibrillating the hamster.

rbergen, 2026-01-03:
You're welcome!

I did some additional desk research, and found that the stock service proxy we were running was an outdated one. It didn't handle a somewhat common error condition that I addressed in the code in November '24, but that fix never trickled down to what was running.

That fixed version is the version that is running now, so I am more confident that we will have continued stock effect service from now on.

rbergen, 2026-01-03:
Man, that literally gave me the shivers...