PlummersSoftwareLLC/NightDriverStrip #645: connection refused error on web server

fEdlbV9IdW50ZXJ8, 2024-07-21:
## Bug report
console shows WL_CONNECTED, but connecting to 192.168.1.71 in a browser throws "connection refused" error

**Problem**
after installing ledstrip v040 on my esp32 [esp-WROOM-32 chip on a ESP32 DEVKITV1 board], and attempting to connect it to wifi, it says "timeout", but console shows WL_CONNECTED with IP 192.168.1.71. When I try to connect to 192.168.1.71 in a browser, it shows "connection refused", and if I try with port 49152, it says "connection reset". I expected to see some type of web UI?

**Steps**

1.  Install ledstrip v040 on an esp32 DEVKIT1 board, selecting esp32 and ledstrip from the dropdown menu
2.  configure wifi
3.  connect to 192.168.1.71 in a chrome browser
4. browser shows "connection refused"

**Notes**
This is my first time using an ESP32, and first time submitting a Github issue, so I am likely forgetting something that would obvious to more experienced users. Be nice.
[esp-web-tools-logs.txt](https://github.com/user-attachments/files/16325324/esp-web-tools-logs.txt)

rbergen, 2024-07-21:
This is because on the project you selected, the web server isn't enabled. This is indicated by the W not being included in the project's feature letters. 

In fact, none of the ESP32 projects have the web server enabled.

fEdlbV9IdW50ZXJ8, 2024-07-21:
so, what is the socket server for then? why does this need to connect to the internet?

fEdlbV9IdW50ZXJ8, 2024-07-21:
> This is because on the project you selected, the web server isn't enabled. This is indicated by the W not being included in the project's feature letters.
> 
> In fact, none of the ESP32 projects have the web server enabled.

so, what is the socket server for then? why does this need to connect to the internet?

rbergen, 2024-07-21:
The socket server can be used to send color data to the ESP32 for display on the LEDs connected to it.

I think all of this is covered to some extent in de README.

If you have no use for that, then there is indeed no need to connect it to WiFi.

fEdlbV9IdW50ZXJ8, 2024-07-21:
operator error: did not realize you have to send packets to the IP, not open it in a browser. [time to learn how to to that XD]

closed 2024-07-21 by fEdlbV9IdW50ZXJ8