PlummersSoftwareLLC/NightDriverStrip #662: Tweak StockTicker effect

davepl, 2024-11-02:
The stock ticker effect currently has all the data it needs to display a proper stock quote, but the change is not displayed correctly.

The intent of the effect is to show CHANGE SINCE LAST CLOSE.  So the code needs to likely do something like look for the last price on a prior calendar day to determine the last close price (or it may be in the data).

The little graphic it displays should also be an indication of the total change from last close to now (though it only shows the most recent 64 data points).

If you go to Google and search for "MSFT stock quote", it will show you the correct delta that we should be showing, give or take 20-30 minutes of activity (our quotes are a bit delayed).

rbergen, 2024-11-03:
Are we sure the effect has all the data it needs? One thing I notice is that a request for the MSFT quote from the stock data server (davepl:8888) currently yields data points that are one hour apart, ranging from October 24th to October 30th, 4PM. I'm pretty sure stock was traded until EOB on November 1st - Google certainly says so -, so it seems to me some data is missing.

Also, if we want the effect to show the change since last close, shouldn't the "last close" value be included in an easily accessible way (i.e. as an attribute) in the response from the stock data server?

davepl, 2024-11-03:
I should say that I *assume* all of the data points needed are there - if it turns out the stock server needs to return something different or additional to what it does now, let me know, and I can make that change or set up Robert or whomever works on it with the setup needed to tweak it directly!

Cheers,
Dave

rbergen, 2024-11-03:
I'm actually very happy to look into the stock server side of things myself (been there, done that, so have everything local setup already). I was just making the point that we should be careful not to assume the right place to act is here (i.e. the effect).

I'll dig into what goes into and comes out of the stock server tomorrow, and I'll report back here with what I make of that.

rbergen, 2024-11-04:
With @davepl's kind cooperation I've made some changes to the server that the stock ticker effect gets its stock data from, so it now also includes the previous day's close price for a stock. That price is in the new `previous_close` property in the server's response JSON object, which is not yet used by the StockTicker effect. Note that this property may be `null` if the previous day's close price is not available.

@robertlipe With this property added, I think this issue is actually ready to be picked up at the StockTicker effect side, if you're still interested in doing so.

closed 2024-11-29 by davepl

rbergen, 2024-11-30:
You really just started looking at this? That's quite the coincidence!

I had a moment yesterday to think about what the changes would have to be, and once I had that clear (and they were quite manageable) I concluded the best way to remember was to just type it into the code.

I apologize for any time you feel you wasted on investigating because of this.

rbergen, 2024-11-30:
Ah, ok. I literally picked up the effect side of this for the first time yesterday. All the work I did before that was on the stockserver proxy.

PR #670 (2024-11-29): Make Stock effect compare against previous day's close
## Description

This closes #662 by making the effect compare against the previous day's close instead of today's open. It does this both in the numeric change shown, as in the graph underneath it.

There is still one "peculiarity" that people may notice, which is that the data points in the graph include after-hour trading, and we don't show this in the coloring of the graph (Google and other sources tend to show after-hour trading in a different color).

This means that the last data point for a calendar day may be positive (green) while [... cut]