PlummersSoftwareLLC/NightDriverStrip #666: Socket server: Unknown command in packed received: [some number]

jmadotgg, 2024-11-09:
## Bug report

**Problem**

I get the error "Unknown command in packet received: 31202" when running the ledstrip example with the python audioserver.py sample script. After that the connection is closing and the server as well as the python client keep retrying to establish a connectionn. I've looked into the socket server and the command16 variable is only supposed to have the values 3 or 4, is that right?
**Steps**

1.  Install python audioserver.py dependencies, setup correctly with esp32 ip.
2.  compile ledstrip example with correct ssid and password in secrets.h
3.  Upload example and run python script

**Example**
Is something here not configured correctly?
```python
#!/usr/bin/python
##+--------------------------------------------------------------------------
##
## audioserver - (c) 2023 Dave Plummer.  All Rights Reserved.
##
## File:        audioserver.py - Pushes Audio FFT data to NightDriverStrip
##
## Description:
##
##    Samples the system audio, splits it into bands using an FFT, and then
##    sends is over WiFi to a NightDriverStrip instance
##
## History:     Feb-20-2023     davepl      Created
##
##---------------------------------------------------------------------------

import pyaudio
import numpy as np
import socket
import struct
import time
import sys

# NightDriver ESP32 wifi address - update to your ESP32 WiFi

client = '192.168.178.72'        

# Set up audio input stream. 512@24000 gives a nice framerate.  And 512
# is what I run on the ESP32 if connected via hardware mic, so at least it matches

chunk_size   = 512
sample_rate  = 44100
max_freq     = 20000
num_bands    = 12

# Correction I apply to get a mostly linear response across the bands.  

if num_bands == 16:
    band_scalars = [ 0.35, 0.20, 0.125, 0.1, 0.5, 1.2, 1.7, 2.0, 2.1, 2.75, 2.0, 8.0, 8.0, 8.0, 8.0, 8.0 ]
else:
    if num_bands == 12:
        band_scalars = [ 1.0, 1.0, 1.0, 1.0, 0.01, 0.01, 0.01, 0.1, 0.1, 0.1, 0.1, 1.0 ]

# Open the audio stream.  I'm reading from the mic here because I could not find a system independent
# way to read from the default output device, which would normally require a loopback instance be 
# installed and for simplicity sake, I don't want that.

p = pyaudio.PyAudio()

#
# Set up FFT parameters:
#

fft_size = 2**12  # Choose the size of the FFT (power of 2 for efficiency)

# Calculate the frequencies corresponding to each FFT bin.  

freqs = np.fft.rfftfreq(fft_size, d=1.0/sample_rate)  

# Divide the frequency range into frequency bands of equal logrithmic width
# `20` is the minimum frequency of human hearing, `max_freq` is the maximum frequency of interest, 
# and `num_bands` is the desired number of frequency bands.  

bands = np.logspace(np.log10(20), np.log10(max_freq), num_bands+1).astype(int)  

# Compute the width of each frequency band.  This returns the detla between band (n, n+1)) for each band

band_widths = np.diff(bands)  # Take the difference between adjacent frequency band limits

# The socket we will open to the ESP32 to send our band data to

print("Connect to " + client + "...")

sock = None

stream = p.open(format=pyaudio.paFloat32, channels=1, rate=sample_rate, input=True, frames_per_buffer=chunk_size)

# Loop to continuously sample audio and compute spectrum analyzer bands
while True:

    # Connect to the socket we will be sending to if its not already connected
    if sock == None:
        sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        address = (client, 49152)
        sock.connect(address)
        sock.setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)
        sock.setblocking(True);

    # Read the raw audio data.  We ignore overflow exceptions from not accepting every bit, it's ok if
    # miss a few in betweeen samples
    audio_data = np.frombuffer(stream.read(chunk_size, exception_on_overflow=False), dtype=np.float32)
    
    # Compute the FFT to put the samples into frequency
    fft_data = np.abs(np.fft.rfft(audio_data, n=fft_size))

    # Compute band values
    band_values = []
    for i in range(len(bands)-1):                                       # BUGBUG RANGE stops one before
        band_start = np.searchsorted(freqs, bands[i])
        band_stop = np.searchsorted(freqs, bands[i+1])
        band_value = np.median(fft_data[band_start:band_stop])
        band_values.append(band_value)
    
    band_values = np.multiply(band_values, band_scalars)

    # Scale band values to [0, 1]
    band_values = np.clip(band_values, 0.000001, None)                  # Avoid div by zero
    max_band_value = np.max(band_values) + 0.000001;                    # Avoid zero maximumum
    if (max_band_value > 1):
        scaled_values = np.divide(band_values, max_band_value)
    else:
        scaled_values = band_values
        
    # Convert scaled values to ASCII bargraph
    bargraphs = []
    for scaled_value in scaled_values:
        asterisks = "*" * int(round(scaled_value * 8))
        bargraph = f"{asterisks:<8}"
        bargraphs.append(bargraph)

    # Print ASCII bargraphs
    #breakpoint()
    print(bargraphs)
    sys.stdout.write('*')

    # Compose and send the PEAKDATA packet to be sent to the ESP32 NightDriverStrip instance

    packed_data = struct.pack('f' * len(scaled_values), *scaled_values)
    command = 4
    length32 = 4 * num_bands
    seconds = int(time.time())
    micros  = time.perf_counter() - seconds
    
    header1 = (command).to_bytes(2, byteorder='little')             # Offset 0, command16
    header2 = (num_bands).to_bytes(2, byteorder='little')           # Offset 2, num_bands
    header3 = (length32).to_bytes(4, byteorder='little')            # Offset 4, length32 
    header4 = (seconds).to_bytes(8, byteorder='little')             # Offset 8, seconds
    header5 = struct.pack('d', micros)                              # Offset 16, micros
   
    complete_packet = header1 + header2 + header3 + header4 + header5 + packed_data

    try:
        sock.send(complete_packet)
    except socket.error as e:
        #breakpoint()
        print("Socket error!");
        sock.close()
        sock = None
    
    time.sleep(0.015);

sock.close()
```

**Notes**
I have connected to led strips to port 5 of my esp32.

Thank you very much for your help! :)

jmadotgg, 2024-11-17:
I cannot seem to figure out where the error is.

jmadotgg, 2024-11-17:
I cannot seem to get this to work. When I do not set ENABLE_AUDIO to 1 it does not process the peak_data, right? Hence the example audioserver.py won't work? But if I do set it to to 1 it tries to read from a non existing mic. What is the correct way to continue further? Thank you in advance.

jmadotgg, 2025-01-12:
@robertlipe Thank you for your detailed elaboration, but my goal is to only send and receive the data via a socket to the esp32.

Cyborg-Squirrel, 2025-07-05:
I'm running into this error when I open a WebSocket, but a different number. It happens when running my Kotlin code or when I attempt to open a WebSocket using default settings in Postman.

Logs
```
(W) (ProcessIncomingConnectionsLoop)(C1) Unknown command in packet received: 17735
(W) 
(W) (SocketServerTaskEntry)(C1) Socket connection closed.  Retrying...
```

Kotlin code sample
```
val uri = UriBuilder.of(websocketAddress).port(clientEntity.wsPort!!).build()
val future = CompletableFuture<LedStripWebSocketClient>()
val clientPublisher = webSocketClient.connect(LedStripWebSocketClient::class.java, uri)
clientPublisher.subscribe(object : Subscriber<LedStripWebSocketClient> {
    override fun onSubscribe(s: Subscription?) {
      s?.request(1)
    }

  override fun onError(t: Throwable?) {
    state = WebSocketState.DisconnectedIdle
    future.completeExceptionally(t)
  }

  override fun onComplete() {}

  override fun onNext(client: LedStripWebSocketClient?) {
    state = WebSocketState.ConnectedIdle
    future.complete(client)
  }
})

client = future.get(5, TimeUnit.SECONDS)
```

Cyborg-Squirrel, 2025-07-05:
@rbergen thank you for the quick response!

The code snippet is from [this class](https://github.com/Cyborg-Squirrel/color-composer/blob/main/src/main/kotlin/io/cyborgsquirrel/lighting/job/WebSocketJob.kt#L269)

Here are the values stored in the clients table in the Postgres database to give you an example of the values loaded into the LedStripClientEntity used for the WebSocketJob.

[image]

The goal of the WebSocketJob class is to connect to stream color data to the ESP32.

rbergen, 2025-07-05:
I think the repo link you sent me points to a private repo, because GitHub won't allow me access to it.

In any case, the table you included in your last message seems to show you're trying to set up a web socket connection to port 49152. That's not going to work. In short:

- Port 49152 will accept color data (and associated commands), but _only_ in raw network packets sent to a "raw" TCP port. There is no way to send color data _to_ the ESP32 via a web socket.
- It is possible to set up a web socket with the ESP32 on-board web server on port 80, which will allow you to _retrieve_ color data that the ESP32 is showing on the connected LEDs. Again, this is color data coming _from_ the ESP32, not the other way around.

I hope this clarifies why what you're trying is not working, and won't.

aien-dev, 2026-10-05:
A note on the number: 31202 is 0x79E2, so the bytes `E2 79` were read where a command id belongs. That is consistent with the receiver having lost frame alignment partway through a PEAKDATA frame: in builds with audio disabled, before commit d73d18db, the PEAKDATA branch of `socketserver.cpp` read only the 24 byte header and left the audio payload in the stream, so the next "header" was payload bytes. d73d18db now reads and discards the declared payload. (Whether those two bytes were the low bits of the first band value can't be confirmed from the report.)

I built the real `socketserver.cpp` on a Linux host (stubbed platform layer, loopback TCP) and fed it audioserver.py-shaped frames: before d73d18db it reports "Unknown command in packet received: 31202" for a frame whose first float ends in those bytes; at d73d18db and on current main every frame is consumed, with audio enabled or disabled. This was not run on a physical ESP32 and does not cover lwIP behaviour.

Two things to check on your side even with a newer build:
1. With `ENABLE_AUDIO 0` (as in the posted `LEDSTRIP` config) the firmware accepts and discards the peak data, so the sample cannot drive anything. With `ENABLE_AUDIO 1` the band count in the script has to equal the firmware's `NUM_BANDS`. The script you pasted has `num_bands = 12`; the copy in the repository has 16, which matches the default. A 12-band sender is rejected as "Invalid peak data packet".
2. I have not looked at how remote peak data behaves when no microphone is attached.

AI assistance: this analysis was done with the help of an AI coding assistant (Claude); I reviewed it.