PlummersSoftwareLLC/NightDriverStrip #725: Create endpoint to expose device time

Cyborg-Squirrel, 2025-06-28:
Streaming RGB data to Nightdriver can include a timestamp.

To ensure the current timestamp from the host doesn't end up being in the past, or otherwise de-synced, relative to the ESP32's time an endpoint to request the device time would be helpful.

Example function in webserver.cpp

```
void CWebServer::GetTime(AsyncWebServerRequest *pRequest)
{
    debugV("Getting current time...");

    timeval tv;
    gettimeofday(&tv, nullptr);

    auto response = std::make_unique<AsyncJsonResponse>();
    auto jsonObject = response->getRoot().to<JsonObject>();

    auto jsonDoc = CreateJsonDocument();
    jsonDoc["seconds"] = tv.tv_sec;
    jsonDoc["microseconds"] = tv.tv_usec;

    jsonObject.set(jsonDoc.as<JsonObjectConst>());

    AddCORSHeaderAndSendResponse(pRequest, response.release());
}
```

Cyborg-Squirrel, 2025-06-28:
There are two problems I'd like to solve by querying the current time of the ESP32. I wouldn't need a set time endpoint and I get that could lead to a lot of wider design problems if a client could arbitrarily set the ESP32's time.

1. If I wanted to run Nightdriver clients on a private network without internet access would there be another way to keep the timestamp in sync?
2. Responsiveness. Getting the current ESP32 device time can give the PC streaming to Nightdriver an idea of network latency if both are using NTP (which I have configured). If I stream color data timestamped at the current time plus a few milliseconds that might work most of the time but, over WiFi this could result in that timestamp being in the past when it is received. Being able to add less time to the timestamp makes the process more responsive when I want it to start or stop.

rbergen, 2025-06-28:
I'm going to agree with @robertlipe on this one. If time sync between ESP32 device and anything else is important, they should both get their time from a reliable source using a proven protocol to align with that source, and that is NTP. It's also how the time syncing challenge has been addressed for a long time in every environment where an (Inter)network connection is available.

Also, if you were to establish that an ESP32 device with NTP enabled does show meaningful time drift, then we should really focus on fixing that.

I'll convert this Issue to a Discussion for house keeping reasons.