PlummersSoftwareLLC/NightDriverStrip #729: Node Package Rollup parsAst.js module not found

JasonDeGroff, 2025-07-16:
Below is the error I received that preventing build from fresh install of platformIO, nodejs, and a fresh opening of this project.

Error [ERR_MODULE_NOT_FOUND]: Cannot find module '.\NightDriverStrip-main\site\node_modules\rollup\parseAst' imported from .\NightDriverStrip-main\site\node_modules\vite\dist\node\cli.js

I was able to get it to resolve by moving parseAst.js from the .\main\site\node_modules\rollup\dist location  up one file to .\main\site\node_modules\rollup. This allowed the project to build using pio run -e demo. After the build I noticed that The rollup directory had way more stuff in it than before.

I am not sure this is fixable in this project, and may be a result of the node package, or my own ignorance, but I though I would report the error and solution I found regardless

rbergen, 2025-07-16:
Thanks for flagging this.

I am inclined to convert this issue to a discussion, as I can't reproduce this locally and a fresh CI build (where all configurations are built with a fresh install of everything) succeeded. 

I did find this issue in the "rollup" package, but that is unrelated to this project as such: https://github.com/rollup/rollup/issues/5199.

Before I make a final decision about this, can you provide some more information about your setup? OS and version, PlatformIO version, NodeJS version, etc.

JasonDeGroff, 2025-07-16:
Sure thing. 
Nodejs v22.17.1
PlatformIO Core 6.1.18
OS: Windows 11 10.0.26100 Build 26100

I installed the PlatformIO extension and nodejs just earlier today. I did fail to install nodejs before trying to build the first time and went and did that, and had to add node to the system path variables.

rbergen, 2025-08-29:
@JasonDeGroff My apologies for not following up on this earlier. Are you still experiencing this problem?