PlummersSoftwareLLC/NightDriverStrip #768: Does polarmap actually required HUB75 functionality?  If not, move to GFXBase

davepl, 2025-08-28:
I'm wondering if there's any reason that the polar map coords, and the effects that use them, must be inextricably bound to LEDMatrixGFX.

LEDMatrixGFX and LEDStripGFX are hardware abstractions that allow effects to run on either, and both support everything all the way down to setpixel.  So I'm thinking this functionality could be moved to the GFXBase class.

That would allow it to be used by Matrices AND strips that are arranged into matrices, without having any hardware dependencies on the HUB75 side.

Unless there are HUB75/Smartmatrix dependencies.... but I didn't spot any in a quick check!

rbergen, 2025-08-29:
> Can we add Robert as someone we can assign issues to, or did I just miss how to do it?

I've just invited him as a Triage collaborator to this repo. That should do the trick.

rbergen, 2025-08-29:
> Actually, can Robert just be trusted to assign, move, and close issues?

:) The Triage role I issued you before you asked should allow that, and a few other things.

rbergen, 2025-08-30:
I know this suggestion would lead to a PR with a LOT of small code changes, but shouldn't we just rename `LEDStripGFX` to `FastLEDGFX` then, and `LEDMatrixGFX` to `SmartmatrixGFX`?

I'm asking this because you state the implementations are hardware-bound, but what I'm then reading is that they're _really_ library-bound and the code I'm looking at - other than what was added in #757 - actually confirms that.

davepl, 2025-08-30:
> rbergen

LEDStripGfx and LEDMatrixGfx make sense to me, but that may just be historical.  If you think there is clearer naming, no objection to doing so.

-Dave

rbergen, 2025-08-31:
At the risk over over-explaining: I opened a draft PR yesterday (#771, draft for reasons indicated in the PR description) to "visualize in code" the implementation of the topic of this issue and a rename of the LED____Gfx classes in accordance with the suggestion I made a few comments up. 

Just noting it this way in case the GitHub "this issue was mentioned in that PR" link gets overlooked.

closed 2025-09-04 by rbergen